drom/circuit #3: Symbols collection

drom, 2015-12-15:
https://commons.wikimedia.org/wiki/File:Logic-gate-index.png

[image]

https://commons.wikimedia.org/wiki/File:2-MUX_Aufbau2_DIN40900.svg

[image]

https://commons.wikimedia.org/wiki/File:3MUX-Primzahl-Aufbau3_DIN40900.svg

[image]

https://commons.wikimedia.org/wiki/File:3MUX-Primzahl-Aufbau2_DIN40900.svg

[image]

https://commons.wikimedia.org/wiki/File:2-MUX_Aufbau_DIN40900.svg

[image]

DmitryAGusev, 2016-02-03:
For multiplexer, some simple view like https://en.wikipedia.org/wiki/Multiplexer#/media/File:Multiplexer_2-to-1.svg should be supported.

drom, 2016-02-03:
@DmitryAGusev , agree, `mux` and `ff` are two next obvious candidates to support. One thing they bring is: heterogeneous pin structure.

`mux` has normal data inputs, selector (one or many), output.
`ff` has data input, clock, enable, reset, set, output (may be inverted output?).

All that need a better format to express it: https://github.com/drom/logidrom/issues/4

drom, 2016-03-13:
Multiplexer related issue: #10