drom/circuit #6: Backward connection

DmitryAGusev, 2016-02-03:
Is it possible to draw backward connections, for loops, or logic with triggers? I.e. https://en.wikipedia.org/wiki/Flip-flop_(electronics)#/media/File:SR_(Clocked)_Flip-flop_Diagram.svg

drom, 2016-02-03:
@DmitryAGusev this is the next big step for the project. Currently, only the forest of tree-like structures is supported. You can visually trace connectivity by signal name, but it isn't visually appealing, agree.

One of the problems, I think, is to express your intent in the input language, and being able to control the layout and routing of the rendering process.

I am working on the AST format that would be capable expressing nested connectivity here: https://github.com/drom/logidrom/blob/master/ref/ast.md

Any help, suggestions, ideas, comments are welcome!

DmitryAGusev, 2016-02-04:
I'm looking for some visualization help in critical path analysis. I think I'll be able to write simple script to analyse reports and convert description of a single path into input text structure for your tool. But more interesting thing is to combine several paths into one scheme, especially when some logic is present in many paths. This will help to highlight hot spots. For these language extensions to support multiple outputs is required.
For single path visualization the following things will be useful:
1. Have ability to indicate length (in time units) of the wires and units
2. Use different colors to highlight wires and units

drom, 2016-02-05:
@DmitryAGusev ahh, yes, critical path reports. Simple version sounds possible with current format.
Probably with minor fixes, like `mux`, `dff`, `box`.
If you need to show only single critical path per diagram, with all units / wires involved, then you may not need highlighting.
We have to think how to add / render extra attributes, like extra text on wires and units.

drom, 2016-03-13:
See #10 fore Multiplexer related discussion.

drom, 2016-05-08:
@DmitryAGusev I was thinking about `critical path analysis` tool. Here is my first sketch: https://github.com/drom/sta