drom/circuit #9: Signal fanout

drom, 2016-03-07:
As it was mentioned in this @wavedrom group topic: https://groups.google.com/d/topic/wavedrom/iqTKBLl6mYU/discussion

> Can some tell me how to make a driver gate to drive multiple gates in the next level with replication of the driver gate.
> For example:
> Level 0 : AND GATE 
> Level 1:  3 Inverters driven by the same AND GATE

drom, 2016-03-07:
TODO:
 * [ ] scope all the cases 
 * [ ] sketch the cases
 * [ ] appropriate grid design
 * [ ] routing algorithm

drom, 2016-03-07:
Easy case 1:
```js
{ assign: [
  ["out", ["&", "a", "b", "b"]]
]}
```
Now:
[image]

Expected:
[image]

DmitryAGusev, 2016-03-08:
Comment to the initial question. Is it possible to add unique tag/id/name to the block, or assign a name to the output signal of any block (not only for global output)? If so, task becomes the same as for input signals in examples above.

drom, 2016-03-08:
ok, you are right @DmitryAGusev . My pictures above have not addressed the original question. How about next example as the possible solution?
```js
{ assign: [
  ["a",
    ['|',
      ["&", ["!s0", ["~", "s0"]], ["!s1", ["~", "s1"]], "e0"],
      ["&", "!s0", "s1", "e1"],
      ["&", "s0", "!s1", "e2"],
      ["&", "s0", "s1", "e3"]
    ]
  ]
]}
```
[image][image]

DmitryAGusev, 2016-03-09:
Yes, this is what I meant. It will be nice to add the ability to suppress the showing of intermediate names like !s0, !s1.

DmitryAGusev, 2016-03-09:
Yes, this is what I meant. It will be nice to add the ability to suppress the showing of intermediate names like !s0, !s1. I.e. by using another quotes '!s1' instead of "!s1"

DmitryAGusev, 2016-03-09:
Yes, this is what I meant. It will be nice to add the ability to suppress the showing of intermediate names like !s0, !s1. I.e. by using another quotes '!s1' instead of "!s1". And the same for input signal names, sometimes some of them are not important to be shown, i.e. for highlighting critical paths.

drom, 2016-03-09:
Another important question is: How to render connect between multiple expressions?
Gray to Binary example:
```js
{ assign:[
  ["b3", "g3"],
  ["b2", ["^", "b3", "g2"]],
  ["b1", ["^", "b2", "g1"]],
  ["b0", ["^", "b1", "g0"]]
]}
```
[image]

DmitryAGusev, 2016-03-10:
It looks like the structure {assign: [ ]} should define some kind of name space. All signals within this space with similar names should be connected (unless it is suppressed explicitly by some option/flag). Two output signals with similar names cause an error. Multiple intermediate inputs can be connected to one intermediate output signal or one global input signal.

drom, 2016-03-10:
That is correct. We have to think about the temporal variable scoping. Especially, if we going to use multiple scopes (namespaces), or / and nested scoping.