drom/circuit #15: How to widen the render area in schematic?

CacahueteNC, 2020-04-23:
I've been using Wavedrom for a while in my docs and I stumbled upon a case where I thought that a logic diagram would convey more meaning than a written expression. And I was correct and was able to render the function in no time.

The problem was that my signal names are not single letters and they overflow the rendering area. I thought that it was just a render issue and exported the PNG but it was the same. 

Here is the [sample in the browser](https://wavedrom.com/editor.html?%7B%20assign%3A%5B%0A%20%20%5B%22Out%20Signal%22%2C%0A%20%20%20%20%5B%22%7C%22%2C%0A%20%20%20%20%20%20%5B%22%26%22%2C%20%5B%22~%22%2C%20%22First%20Input%22%5D%2C%20%22Good%20signal%22%5D%2C%0A%20%20%20%20%20%20%5B%22%26%22%2C%20%5B%22~%22%2C%20%22Second%20Input%22%5D%2C%20%22Great%20signal%22%5D%0A%20%20%20%20%5D%0A%20%20%5D%0A%5D%7D) and here is the PNG result:
```
{ assign:[
  ["Out Signal",
    ["|",
      ["&", ["~", "First Input"], "Good signal"],
      ["&", ["~", "Second Input"], "Great signal"]
    ]
  ]
]}
```
[image]

I wanted to export it as SVG and have it scale correctly in Microsoft Visio, but the later does not seem to be happy with it, here's what I see :

[image]

Is there a way to enlarge the rendering area, something like `config: { hscale: 2 }`?

Thanks a lot for your time!

drom, 2020-04-23:
related to #30

drom, 2020-04-24:
It should be done automatically. I have the facility to measure test block with for specific font/size. But will require a bit more thinking.